monash-automation/mes-printing-server #48: feat: filament management

jichengzhi, 2024-04-07:
Need APIs to manage filaments in the lab. For now, each step is done manually. for example:

- user add record (PLA, yellow, 1m) after taking certain amount of filament
- admin check records and current inventory
- admin add new filaments to inventory

jichengzhi, 2024-04-08:
# Manually Logging from Keenan:

## Filament Stock
[image]

## Filament Loading/Unloading
[image]

## Print Jobs
[image]

shiki65536, 2024-04-22:
#### Update:
- Created branch filament to address the issue

#### Questions:
- Not sure the following feature is about "Update" or "Create" another table to record.
- Is filament transaction needed?

#### Issue:
- Struggling in writing test_filament.py

jichengzhi, 2024-04-22:
> #### Update:
> * Created branch filament to address the issue
> 
> #### Questions:
> * Not sure the following feature is about "Update" or "Create" another table to record.
> * Is filament transaction needed?
> 
> #### Issue:
> * Struggling in writing test_filament.py

For future commits, submit a PR instead of pushing a new branch, otherwise I cannot write reviews to your code

shiki65536, 2024-04-22:
> > #### Update:
> > 
> > * Created branch filament to address the issue
> > 
> > #### Questions:
> > 
> > * Not sure the following feature is about "Update" or "Create" another table to record.
> > * Is filament transaction needed?
> > 
> > #### Issue:
> > 
> > * Struggling in writing test_filament.py
> 
> For future commits, submit a PR instead of pushing a new branch, otherwise I cannot write reviews to your code.
> 
> An example is #33.

I am so sorry, totally forgot I should fork then PR.
I've forked and updated the code in my repo. We can delete the branch here.
Apologize again for how careless I am.

jichengzhi, 2024-04-24:
Close due to inactivity for more than 2 weeks

closed 2024-04-24 by jichengzhi